OpenFAST/r-test #16: Out of date TurbSim input files

andrew-platt, 2019-06-14:
The input files for `TurbSim` given in `https://github.com/OpenFAST/r-test/tree/dev/glue-codes/openfast/5MW_Baseline/Wind` are out of date (version 1.5 whereas `TurbSim` is at version 2.x.

andrew-platt, 2019-06-14:
This file works with `TurbSim` from  2d9230d 
[TurbSim_2.0.txt](https://github.com/OpenFAST/r-test/files/3292412/TurbSim_2.0.txt)

ewquon, 2020-11-18:
I can confirm that this issue still exists and @andrew-platt's sample input file still works with the turbsim module in openfast v2.4 (https://github.com/OpenFAST/openfast/commit/8fabb647332548fa682644f473a342cafc1c9c55)

bjonkman, 2022-01-12:
I updated the TurbSim input files for (hopefully) all of the TurbSim files in the current r-test in this branch: https://github.com/OpenFAST/r-test/tree/f/TurbSim_mods; They work with TurbSim changes in OpenFAST's pull request https://github.com/OpenFAST/openfast/pull/887

closed 2022-10-25 by ebranlard