OpenFAST/r-test #19: .fst input files

tchatte3, 2020-05-04:
In the case-specific directories inside r-test/glue-codes/openfast/ I cannot find .fst files. Any help? Or am I looking at the wrong directories?

jjonkman, 2020-05-04:
Dear @tchatte3,

I'm not sure I understand your question.  Most of the directories in r-test/glue-codes/openfast/, e.g., r-test/glue-codes/openfast/5MW_ITIBarge_DLL_WTurb_WavesIrr include a *.fst file.

Best regards,

tchatte3, 2020-05-04:
Dear @jjonkman, 

thank you for the quick response. Yes, some of the directories do contain .fst files, but some do not as well. Say, in 5MW_Baseline/ and in AWT_27/  I could not find the .fst files. I was specifically interested in some of these .fst files when for example, 5MW_Baseline is getting the input from TurbSim input file. 

-Tanmoy

jjonkman, 2020-05-04:
Dear @tchatte3,

Several of the directories in r-test/glue-codes/openfast/ --i.e., 5MW_Baseline, AOC, AWT27, Ideal_Beam, SWRT, UAE_VI, and WP_Baseline--contain common module-level input files used by several of the regression tests; these directories do not contain *.fst files.

Best regards,

tchatte3, 2020-05-04:
Dear @jjonkman . I understand now. Thanks for the nice explanation.

-Tanmoy

closed 2020-05-04 by jjonkman