OpenFAST/r-test #28: Error: Executable does not have proper permissions

TiasaGhosh, 2020-07-23:
Hello everybody!
I am new to OpenFAST and have been trying to run some regression tests. I am using Cygwin method to simulate the cases.
I trying to run the executable 'executeOpenfastRegressionCase.py ' with Python3.
when I am entering all the required arguments, I am getting an error as the attached  screenshot.

How do I solve this? Any help is appreciated 👍 
Thanks!
[image]

rafmudaf, 2020-07-23:
I think you need the other slash in Cygwin (try `/` instead of `\`).

TiasaGhosh, 2020-07-24:
Hi rafmudaf,
thank you for the response. It worked actually. :) 
But I ran into some other issue now :/ 
[image]

am I missing something? 
Thanks again! :)

rafmudaf, 2020-07-24:
I think you're trying to run the Windows binary on Cygwin. Its subtle, but the Cygwin environment is not the same as a Windows. I dont think the binaries that we distribute through GitHub will run in Cygwin.

If you dont need to modify the source code, you could run the regression tests in Windows command prompt or PowerShell with the binaries from GitHub.
If you _do_ need to modify the code or otherwise need to run in Cygwin, you'll have to compile it from scratch there. See the documentation on [compiling or linux with cmake](https://openfast.readthedocs.io/en/master/source/install/index.html#cmake-with-make-for-linux-macos).
Alternatively, if you want linux on Windows, I highly recommend using the Windows Subsystem for Linux.

closed 2020-07-31 by rafmudaf