OpenFAST/r-test #34: Path to DISCON.dll library

jwpeterson, 2020-10-20:
Hi,

I tried running the 5MW_Land_BD_DLL_WTurb.fst input file and it died with an error:
```
FAST_InitializeAll:SrvD_Init:BladedInterface_Init:The dynamic library
./../5MW_Baseline/ServoData/DISCON.dll could not be loaded. Check that the file exists in the
specified location and that it is compiled for 64-bit applications.
FAST encountered an error during module initialization.
 Simulation error level: FATAL ERROR
 Aborting OpenFAST.
```

I found that if I change the DLL_FileName path to include "build", then it runs successfully. 

```patch
diff --git a/glue-codes/openfast/5MW_Land_BD_DLL_WTurb/NRELOffshrBsline5MW_Onshore_ServoDyn.dat b/glue-codes/openfast/5MW_Land_BD_DLL_WTurb/NRELOffshrBsline5MW_Onshore_ServoDyn.dat
index 6083fef..46d20a9 100644
--- a/glue-codes/openfast/5MW_Land_BD_DLL_WTurb/NRELOffshrBsline5MW_Onshore_ServoDyn.dat
+++ b/glue-codes/openfast/5MW_Land_BD_DLL_WTurb/NRELOffshrBsline5MW_Onshore_ServoDyn.dat
@@ -63,7 +63,7 @@ False         CompNTMD     - Compute nacelle tuned mass damper {true/false} (fla
 False         CompTTMD     - Compute tower tuned mass damper {true/false} (flag)
 "../5MW_Baseline/NRELOffshrBsline5MW_ServoDyn_TMD.dat"    TTMDfile     - Name of the file for tower tuned mass damper (quoted string) [unused when CompTTMD is false]
 ---------------------- BLADED INTERFACE ---------------------------------------- [used only with Bladed Interface]
-"../5MW_Baseline/ServoData/DISCON.dll"    DLL_FileName - Name/location of the dynamic library {.dll [Windows] or .so [Linux]} in the Bladed-DLL format (-) [used only with Bladed Interface]
+"../5MW_Baseline/ServoData/DISCON/build/DISCON.dll"    DLL_FileName - Name/location of the dynamic library {.dll [Windows] or .so [Linux]} in the Bladed-DLL format (-) [used only with Bladed Interface]^M
 "DISCON.IN"    DLL_InFile   - Name of input file sent to the DLL (-) [used only with Bladed Interface]
 "DISCON"      DLL_ProcName - Name of procedure in DLL to be called (-) [case sensitive; used only with DLL Interface]
 "default"     DLL_DT       - Communication interval for dynamic library (s) (or "default") [used only with Bladed Interface]
```

rafmudaf, 2020-10-27:
Did you run the `make install` command? If so, it will place all of the input files, including the DISCON library, in the correct place. Otherwise, your workaround is fine, too.

jwpeterson, 2020-10-27:
I think I followed the [instructions for building unit tests](https://openfast.readthedocs.io/en/master/source/testing/unit_test.html) which just said to pass
```
-DBUILD_TESTING=ON
```
when building OpenFAST itself, and then I ran "make install" for OpenFAST itself. Is there a separate "make install" step that one should do for the unit/regression test suites?

rafmudaf, 2020-10-27:
That should do it. So the full process for configuring CMake to build the unit tests and set up the regression tests is:

```
# assuming you're in the openfast/build directory

# Generate the appropriate Makefiles, configure the pFUnit project which is required for the unit tests, and move files around for the regression tests
cmake .. -DBUILD_TESTING=ON

# Compiles all available make-targets and place the binaries in, by default, openfast/install. It also places the three DISCON libraries at openfast/build/reg_tests/glue-codes/5MW_Baseline/ServoData
make install

# See if one of the 5MW offshore cases will run
ctest -VV -R 5MW_ITIBarge_DLL_WTurb_WavesIrr
```

I think CMake on a Mac may make the default install directory `/usr/local`, so check that in your CMakeCache or with `ccmake`.

rafmudaf, 2020-10-27:
@jwpeterson Let me know if that didn't work out for you.

closed 2020-10-27 by rafmudaf

jwpeterson, 2020-11-03:
Thanks @rafmudaf, I undid my diff above and ran
```
ctest -VV -R 5MW_Land_BD_DLL_WTurb
```
This seemed to work fine, so I guess my issue was that I did not use the `ctest` executable to run the test previously? For reference, the command that I tried previously (and which does not work is):  
```
cd $OPENFAST_GIT_REPO/reg_tests/r-test/glue-codes/openfast/5MW_Land_BD_DLL_WTurb
/opt/openfast/bin/openfast 5MW_Land_BD_DLL_WTurb.fst
```
Notes: 
* The openfast git repo is in $OPENFAST_GIT_REPO and 
* I configured openfast with `-DCMAKE_INSTALL_PREFIX="/opt/openfast"`

so I guess that `ctest` sets up some paths, etc. which allows this particular example to work? Or perhaps this input file just isn't meant to be run directly via the openfast executable in this way...

rafmudaf, 2020-11-03:
Ah so your issue is that when you changed the install prefix, you changed the destination for the OpenFAST executable itself and also the DISCON libraries. Since you compiled with static linking (this is the default), it doesn't matter where the OpenFAST executable is so long as you have appropriate permissions to execute it. However, you already saw that they path to the DISCON library in the `.fst` file has a relative path, so the `DISCON.DLL` has to be at a specific place. If you had not changed `CMAKE_INSTALL_PREFIX`, running `make install` would have placed all of the binaries in the correct place for the regression tests.

For what its worth, I've recently added a new regression test make-target that only compiles the OpenFAST executable and places the DISCON libraries in the correct spot regardless of the install path. Pull the dev branch and run `make regression_tests` for this.

rafmudaf, 2020-11-03:
If you inspect the CMake configuration in `openfast/reg_tests`, you'll see that `ctest` isn't actually doing very much. It simply makes a system call to a few Python scripts. The benefits of using ctest are batch execution and the test filtering (regular expressions with `-R`, labels with `-L`), and there's additional functionality that we don't use. I'm just pointing this out to clear up any possible confusion about where your problem was. CTest is a wrapper to another wrapper to the OpenFAST executable. CMake does move files around in addition to creating all of the Makefiles. Ultimately, if something in the regression tests isn't running, it's probably a configuration problem.

jwpeterson, 2020-11-03:
> Since you compiled with static linking (this is the default),

Oh, I forgot to mention that I also configured with `-DBUILD_SHARED_LIBS=ON`, as well as setting `-DCMAKE_INSTALL_RPATH`, but I don't think that's relevant to this particular issue... everything related to building with shared libraries seems to be working fine.

I'll try your dev branch momentarily.

rafmudaf, 2020-11-03:
Just to be clear, use the dev branch from `OpenFAST/openfast` not `rafmudaf/openfast`. My fork is _usually_ up to date but I make no guarantee about that.

jwpeterson, 2020-11-03:
OK, I think that worked. The output of `make regression_tests` for me is:
```
$ make regression_tests
[  0%] Built target DISCON_OC3Hywind
[  1%] Built target versioninfolib
[ 15%] Built target nwtclibs
[ 26%] Built target mapcpplib
[ 27%] Built target maplib
[ 36%] Built target ifwlib
[ 37%] Built target afinfolib
[ 39%] Built target aeroacoustics
[ 40%] Built target uaaerolib
[ 44%] Built target fvwlib
[ 49%] Built target aerodynlib
[ 53%] Built target aerodyn14lib
[ 56%] Built target servodynlib
[ 57%] Built target elastodynlib
[ 59%] Built target beamdynlib
[ 62%] Built target subdynlib
[ 75%] Built target hydrodynlib
[ 76%] Built target orcaflexlib
[ 79%] Built target extptfm_mckflib
[ 81%] Built target openfoamtypeslib
[ 81%] Built target sctypeslib
[ 82%] Built target feamlib
[ 84%] Built target moordynlib
[ 85%] Built target icedynlib
[ 91%] Built target icefloelib
[ 91%] Built target openfast_prelib
[ 92%] Built target foamfastlib
[ 94%] Built target scfastlib
[ 97%] Built target openfast_postlib
[ 97%] Built target openfast
[ 97%] Built target DISCON
[ 98%] Built target DISCON_ITIBarge
Scanning dependencies of target regression_tests
[ 98%] Generating glue-codes/openfast/5MW_Baseline/ServoData/DISCON.dll
[100%] Generating glue-codes/openfast/5MW_Baseline/ServoData/DISCON_ITIBarge.dll
[100%] Generating glue-codes/openfast/5MW_Baseline/ServoData/DISCON_OC3Hywind.dll
```

I can now run e.g.:
```
/opt/openfast/bin/openfast $OPENFAST_GIT_REPO/build/reg_tests/glue-codes/openfast/5MW_Land_BD_DLL_WTurb/5MW_Land_BD_DLL_WTurb.fst
```
successfully, since DISCON.dll's location now matches the relative path given in the input file.