OpenFAST/r-test #35: version 2.4.0 5MW_OC3Mnpl_DLL_WTurb_WavesIrr not working with windows x64 binaries

ManuelaBoehm, 2020-10-23:
Hi Rafael,

I tried to run the r-test 5MW_OC3Mnpl_DLL_WTurb_WavesIrr (v2.4.0) with 2.4.0 windows 64x binaries and I got some errors. There might be an issue with mismatching line numbers in the input files. I didn't have any issues running this r-test with v 2.2.0 and 2.3.0. Please find attached the FAST console output.

[log.txt](https://github.com/OpenFAST/r-test/files/5429059/log.txt)

Best,
Manuela

rafmudaf, 2020-10-23:
Thanks @ManuelaBoehm. For references, here's the content of the log file: 

```
**************************************************************************************************
OpenFAST

Copyright (C) 2020 National Renewable Energy Laboratory
Copyright (C) 2020 Envision Energy USA LTD

This program is licensed under Apache License Version 2.0 and comes with ABSOLUTELY NO WARRANTY.
See the "LICENSE" file distributed with this software for details.
**************************************************************************************************

OpenFAST-v2.4.0
Compile Info:
 - Compiler: Intel(R) Visual Fortran Intel(R) 64 Compiler for applications running on Intel(R)
 64, Version 19.0.5.281 Build 20190815
 - Architecture: 64 bit
 - Precision: single
 - Date: Sep 16 2020
 - Time: 12:05:52
Execution Info:
 - Date: 10/22/2020
 - Time: 09:29:29+0200

OpenFAST input file heading:
    FAST Certification Test #19: NREL 5.0 MW Baseline Wind Turbine with OC3 Monopile RF
    Configuration, for use in offshore analysis

Running ElastoDyn.
Nodal outputs section of ElastoDyn input file not found or improperly formatted.
Running AeroDyn.
Nodal output section of AeroDyn input file not found or improperly formatted.
Running BEM.
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. BladeNode = 1, Blade = 1
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. BladeNode = 2, Blade = 1
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. BladeNode = 3, Blade = 1
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. BladeNode = 4, Blade = 1
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. BladeNode = 1, Blade = 2
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. BladeNode = 2, Blade = 2
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. BladeNode = 3, Blade = 2
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. BladeNode = 4, Blade = 2
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. BladeNode = 1, Blade = 3
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. BladeNode = 2, Blade = 3
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. BladeNode = 3, Blade = 3
Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. BladeNode = 4, Blade = 3
Running InflowWind.

   Reading a 31x31 grid (145 m wide, 17.5 m to 162.5 m above ground) with a characteristic wind
   speed of 12 m/s. This full-field file was generated by TurbSim (v1.06.00, 21-Sep-2012) on
   07-Jan-2014 at 12:50:45.

   Processed 1442 time steps of 20-Hz full-field data (72.05 seconds).
Running ServoDyn.
Running ServoDyn Interface for Bladed Controllers.
Using legacy Bladed DLL interface.
Running HydroDyn.
 Generating incident wave kinematics and current time history.
 Calculating second order difference frequency wave kinematics.
 Calculating second order sum frequency wave kinematics.

FAST_InitializeAll:HydroDyn_Init:MeshCommit: Line2 element 1 has 0 length.
   n2 = n(63) = (0,0,-20)
   n1 = n(1) = (0,0,-20)

 FAST encountered an error during module initialization.
 Simulation error level: FATAL ERROR

 Aborting OpenFAST.
```

rafmudaf, 2020-10-23:
@ManuelaBoehm we are aware of this issue. As a workaround, you can change the node location [here](https://github.com/OpenFAST/r-test/blob/master/glue-codes/openfast/5MW_OC3Mnpl_DLL_WTurb_WavesIrr/NRELOffshrBsline5MW_OC3Monopile_HydroDyn.dat#L98) to `-20.005` or use the double precision executable.

@andrew-platt this is the same problem as in https://wind.nrel.gov/forum/wind/viewtopic.php?f=4&t=1382&start=75
Is the best solution to change the model inputs or change the MESH_COMMIT subroutine to allow for a programmer-defined tolerance?

bjonkman, 2020-10-23:
@rafmudaf , @andrew-platt , I don't think that tolerance should be any smaller than it is now `sqrt(epsilon())`. I updated the mesh mapping to not create new nodes if the elements are smaller than that tolerance when it is augmenting the mesh for line2 mappings, and it's using the same tolerance in the mesh commit routine.

I'd recommend updating the input file(s) so that it doesn't try to create line2 elements with lengths of less than 0.001 m. If you really do need zero-length elements, I think we should make that a special case of the mapping algorithm.

andrew-platt, 2022-11-23:
I believe this has been solved in later releases.  Going to close this issue since it hasn't been updated in 2 years.

closed 2022-11-23 by andrew-platt