OpenFAST/r-test #37: Folder where to clone r-test and pFUnit after installing OpenFAST

Amarelio, 2020-11-26:
Good evening,

I'm new to OpenFAST and I've been able to successfully install it on my Ubuntu machine. The next step would then be to test OpenFAST and for that, as I interpret it from the documentation, I'm supposed to clone the r-test and pFUnit git submodules to somewhere but I cannot understand from the documentation to where I should clone them. Could you let me know?

P.S: There is a similar issue on the openfast "issues" (number 401) but it does NOT precise the folders so the answer is not there. I already read the documentation several times and tried go with my gut but I've reached a dead end so I'm going to reinstall everything again.

rafmudaf, 2020-11-30:
How did you install OpenFAST on your machine?

If you went the usual route of `git clone`, this "tip" in [this section](https://openfast.readthedocs.io/en/master/source/testing/index.html#configuring-the-test-suite) of the docs is what you need.

Amarelio, 2020-11-30:
I went through the usual "git clone". Thanks a lot for the link. Meanwhile, I had already solved my problem. I thought the "git submodule update --init --recursive" command would clone the r-test and pFUnit submodules to the current working directory and that's why I was asking to which folder I should have cloned the submodules. In the end, I ended up typing the command in the command line while being on the OpenFAST folder and it went well.

rafmudaf, 2020-11-30:
Great to hear!

closed 2020-11-30 by rafmudaf