OpenFAST/r-test #39: Add Test case for flexible platforms

andrew-platt, 2021-01-28:
No test cases were added to the regression tests when platform flexibility was added (https://github.com/OpenFAST/openfast/pull/537).

Add a test case using a publicly available structure that has been validated:

1. Simple spar based on OC3
2. paper by Borg et al that includes a model of a flexible spar. This is a model of the DTU 10-MW turbine atop a fictitious spar: https://iopscience.iop.org/article/10.1088/1742-6596/753/8/082024/pdf

jjonkman, 2021-01-28:
Yes, this is definitely something that needs to be developed...just need a project to fund this.  The full-system models we have tested to-date contain proprietary information that cannot be shared publicly.

In addition to the options identified by @andrew-platt, another possibility is the public version of the TetraSpar: https://www.mdpi.com/1996-1073/13/18/4911.

Best regards,

andrew-platt, 2024-09-05:
Tagging @luwang00