OpenFAST/r-test #41: Add test example case of earthquake simulation of jacket structure

andrew-platt, 2021-05-03:
It would be nice to include a simple test case showing a time-series force input at the base of a jacket structure.

This case can be used as a starting point (based on OC4 Jacket, with a 4 Hz force input at base starting at 10 seconds).  This can be run both with and without HD enabled with OpenFAST 3.0.0.

[SeismicExample.zip](https://github.com/OpenFAST/r-test/files/6417976/SeismicExample.zip)

This is related to https://github.com/OpenFAST/openfast/pull/739

L-Ethan, 2022-08-26:
Dear Andrew
Hello! Thank you very much for your sharing. These made me see the possibility of analyzing the response of wind turbine under earthquake in OPENFAST software. For a long time I thought that only the FAST V7 software could perform seismic analysis on wind turbines. But I have a question, cause under the STC module of OPENFAST, the external time-series can only be the load sequence. But the earthquake sequences available on the network are basically acceleration, velocity or displacement sequence. If convenient, would you mind to share your suggestion about how to change the acceleration(velcity\diapalcement) to seismic load?
Best wishes.
LU Yijing

jjonkman, 2022-08-26:
Dear @L-Ethan,

The Seismic functionality of FAST v7 took the specified seismic motion and integrated it to get desired displacement and velocity, which was then converted to a force through a second-order actuator based on a specified frequency and damping.  However, the second-order actuator used in the Seismic functionality computed the force based on the relative difference between the desired and actual motion of the substructure.  The StC functionality of OpenFAST could be used to implement the desired motion leading to a force, but because this force must be fully defined at model initialization, the implementation would miss the contribution to the force from the actual motion of the substructure (which adds stiffness and damping).  You'd have to modify the source code of StC to get this stiffness and damping contribution to fully mimic the Seismic functionality of FAST v7 with the StC functionality of OpenFAST.

See the documentation supplied with the Seismic functionality of FAST v7 for more information: https://www.nrel.gov/docs/fy12osti/53872.pdf.

Best regards,

kashyap5794, 2023-01-17:
Dear Jason, 

What is the best way to implement seismic capability in OpenFAST? 
1. ExtPtfm
2. StC Control Loads

Can you list the limitations briefly of both with regard to seismic implementation in OpenFAST? In addition, is it possible to have prescribed platform motions (due to an earthquake) along with a flexible foundation? 

Thanks
Subham

jjonkman, 2023-01-21:
Dear @kashyap5794,

In my prior post above, I outlined how the StC submodel of the ServoDyn module of OpenFAST could be modified to support functionality similar to the Seismic version of FAST v7.  But this would require a change to the source code.

And the following OpenFAST issue outlines how the external platform (ExtPtfm) module of OpenFAST could be implemented to mimic the Seismic functionality of FAST v7: https://github.com/OpenFAST/openfast/issues/617.

Both of these methods aim at prescribing a motion time history of the rigid foundation (platform) upon which the wind turbine is installed.  By "flexible foundation", presumably you mean modeling the flexibility of a pile with loads distributed along it.  Such functionality has not yet been added to OpenFAST.  That said, the new SoilDyn module to be released in OpenFAST v4.0 provides a good start to that, although the capability is not fully flushed out: https://github.com/OpenFAST/openfast/pull/986.

Best regards,