OpenFAST/r-test #45: The dynamic library  ./../5MW_Baseline/ServoData/DISCON.dll could not be loaded.

AaronCodingG, 2021-11-03:
Hello everyone,

I'm trying to run OpenFAST on an m1 MacBook Air (on MacOS Monterey). But some of the regression tests fail with the following output:
```
(openfast_test) MacBook-Air-2 5MW_Land_DLL_WTurb % openfast 5MW_Land_DLL_WTurb.fst 

 **************************************************************************************************
 OpenFAST

 Copyright (C) 2021 National Renewable Energy Laboratory
 Copyright (C) 2021 Envision Energy USA LTD

 This program is licensed under Apache License Version 2.0 and comes with ABSOLUTELY NO WARRANTY.
 See the "LICENSE" file distributed with this software for details.
 **************************************************************************************************

 OpenFAST-v3.0.0-324-g7a59bec8
 Compile Info:
  - Compiler: GCC version 9.3.0
  - Architecture: 64 bit
  - Precision: single
  - OpenMP: No
  - Date: Sep 30 2021
  - Time: 19:38:10
 Execution Info:
  - Date: 11/03/2021
  - Time: 15:00:26+0100

 OpenFAST input file heading:
     FAST Certification Test #18: NREL 5.0 MW Baseline Wind Turbine (Onshore)

 Running ElastoDyn.
 Nodal outputs section of ElastoDyn input file not found or improperly formatted.
 Running AeroDyn.
 AD15 Nodal Outputs: Nodal output section of AeroDyn input file not found or improperly formatted.
 Skipping nodal outputs.
 Warning: Turning off Unsteady Aerodynamics because UA parameters are not included in airfoil
 (airfoil has likely has constant polars). (node 1, blade 1)
 Warning: Turning off Unsteady Aerodynamics because UA parameters are not included in airfoil
 (airfoil has likely has constant polars). (node 2, blade 1)
 Warning: Turning off Unsteady Aerodynamics because UA parameters are not included in airfoil
 (airfoil has likely has constant polars). (node 3, blade 1)
 Warning: Turning off Unsteady Aerodynamics because UA parameters are not included in airfoil
 (airfoil has likely has constant polars). (node 4, blade 1)
 Warning: Turning off Unsteady Aerodynamics because UA parameters are not included in airfoil
 (airfoil has likely has constant polars). (node 1, blade 2)
 Warning: Turning off Unsteady Aerodynamics because UA parameters are not included in airfoil
 (airfoil has likely has constant polars). (node 2, blade 2)
 Warning: Turning off Unsteady Aerodynamics because UA parameters are not included in airfoil
 (airfoil has likely has constant polars). (node 3, blade 2)
 Warning: Turning off Unsteady Aerodynamics because UA parameters are not included in airfoil
 (airfoil has likely has constant polars). (node 4, blade 2)
 Warning: Turning off Unsteady Aerodynamics because UA parameters are not included in airfoil
 (airfoil has likely has constant polars). (node 1, blade 3)
 Warning: Turning off Unsteady Aerodynamics because UA parameters are not included in airfoil
 (airfoil has likely has constant polars). (node 2, blade 3)
 Warning: Turning off Unsteady Aerodynamics because UA parameters are not included in airfoil
 (airfoil has likely has constant polars). (node 3, blade 3)
 Warning: Turning off Unsteady Aerodynamics because UA parameters are not included in airfoil
 (airfoil has likely has constant polars). (node 4, blade 3)
 Running InflowWind.

    Reading a 31x31 grid (145 m wide, 17.5 m to 162.5 m above ground) with a characteristic wind
    speed of 12 m/s. This full-field file was generated by TurbSim (v1.06.00, 21-Sep-2012) on
    07-Jan-2014 at 12:50:45.

    Processed 1442 time steps of 20-Hz full-field data (72.05 seconds).
 Running ServoDyn.
 Running ServoDyn Interface for Bladed Controllers (using GNU Fortran for Linux, ).

[... 119 lines cut ...]
-- Configuring done
-- Generating done
-- Build files have been written to: r-test/glue-codes/openfast/5MW_Baseline/ServoData/DISCON/build
(openfast_test) MacBook-Air-2 build % make
Scanning dependencies of target DISCON
[ 50%] Building Fortran object CMakeFiles/DISCON.dir/DISCON.F90.o
[100%] Linking Fortran shared library DISCON.dll
ld: warning: dylib (/opt/homebrew/Cellar/gcc/11.2.0/lib/gcc/11/libgfortran.dylib) was built for newer macOS version (11.4) than being linked (11.0)
[100%] Built target DISCON
(openfast_test) MacBook-Air-2 build % ls
CMakeCache.txt          CMakeFiles              DISCON.dll              Makefile                cmake_install.cmake
(openfast_test) MacBook-Air-2 build % cp DISCON.dll ../..
```
The DISCON.dll now lies under along with the other dlls:
'.../r-test/glue-codes/openfast/5MW_Baseline/ServoData/DISCON.dll'

```
(openfast_test) MacBook-Air-2 ServoData % ls
DISCON			DISCON_ITI		DISCON_OC3
DISCON.dll		DISCON_ITIBarge.dll	DISCON_OC3Hywind.dll
```

Now, running the failing tests individually produces the error message at the beginning. I attempted to reproduce the error on Linux (CentOS Linux release 7.4.1708 (Core)) but was not able to reproduce it. I also tried reproducing it on an older x86 MacBook where I was not able to reproduce the issue either. I'm thinking it has something to do with the M1 chip. Does anybody have any suggestions what the issue could be?

Best regards

deslaughter, 2022-11-23:
@AaronTUB I didn't run into this exact issue, but I have a theory about what is happening. 

Is there any chance that your version of Anaconda is for the x86-64 (Intel) processor? If so, it is likely that the OpenFAST Conda environment is also for x86-64 as there isn't currently an arm64 (M1) Conda environment for OpenFAST. Since the shared library was compiled directly on the M1, it is specific to the arm64 processor as was displayed when you ran the `file` command and got `Mach-O 64-bit dynamically linked shared library arm64`. The OpenFAST executable was being run through Rosetta which allowed it to work, but it is not compatible with arm64 shared libraries. This is an unfortunate side effect of the the M1 being a different architecture and it's taken some time to support OpenFAST on the new platform. 

OpenFAST 3.3.0 is available through Homebrew as an arm64 (M1) native executable which should work correctly with shared libraries compiled on arm64.

I'm going to close this issue due its age, but feel free to reopen if this solution doesn't resolve the problem.

closed 2022-11-23 by deslaughter

AaronTUB, 2022-11-24:
@deslaughter thanks a lot for the feedback. That's exactly what the  problem was, now it works perfectly.