OpenFAST/r-test #48: AeroAcoustic files

bjonkman, 2022-01-12:
When comparing some of the input files for the AeroAcoustic module in `IEA_LB_RWT-AeroAcoustics`, I have noticed that some of the boundary-layer input files use "delta" while others use "theta". If I'm reading the documentation and code correctly, these should be using `Delta_SS` and `Delta_PS` instead of `Theta_SS` and `Theta_PS`: 

- [AF05_BL.txt](https://github.com/OpenFAST/r-test/blob/main/glue-codes/openfast/IEA_LB_RWT-AeroAcoustics/Airfoils/AF05_BL.txt)
- [AF06_BL.txt](https://github.com/OpenFAST/r-test/blob/main/glue-codes/openfast/IEA_LB_RWT-AeroAcoustics/Airfoils/AF06_BL.txt)
- [AF07_BL.txt](https://github.com/OpenFAST/r-test/blob/main/glue-codes/openfast/IEA_LB_RWT-AeroAcoustics/Airfoils/AF07_BL.txt)
- [AF08_BL.txt](https://github.com/OpenFAST/r-test/blob/main/glue-codes/openfast/IEA_LB_RWT-AeroAcoustics/Airfoils/AF08_BL.txt)
- [AF09_BL.txt](https://github.com/OpenFAST/r-test/blob/main/glue-codes/openfast/IEA_LB_RWT-AeroAcoustics/Airfoils/AF09_BL.txt)

I am not sure if this is a matter of just the headers having a typo or if the data needs to be converted.

ptrbortolotti, 2022-01-24:
See https://github.com/OpenFAST/r-test/pull/49

ebranlard, 2022-10-25:
This should be fixed with #49 now.