OpenFAST/r-test #58: Error in r-test/glue-codes/openfast/5MW_Land_ModeShapes

Prithvi2029, 2022-05-27:
Hello, 
I was trying to run this case using openfast_x64.exe and 5MW_Land_ModeShapes.fst as the input file but it's showed multiple errors in NRELOffshrBsline5MW_Onshore_AeroDyn15.dat file one of them was missing compAA and after I corrected that keyword, many other errors popped up. Do I need to modify these files before running example test cases or they should be running as it is?

Thank you for your time and guidance.

jjonkman, 2022-05-27:
Dear @Prithvi2029,

I'm not sure why, but it doesn't appear that this r-test input file is up-to-date with OpenFAST v3.1.  However, you should be able to use the AeroDyn v15 input from the following r-test, which is up-to-date, as long as you set `AFAeroMod` = 1 and `FrozenWake` = True: https://github.com/OpenFAST/r-test/blob/main/glue-codes/openfast/5MW_Land_DLL_WTurb/NRELOffshrBsline5MW_Onshore_AeroDyn15.dat.

Best regards,

Prithvi2029, 2022-05-27:
Hello @jjonkman , 
I did try that before and the new error which refers to 5MW Baseline model comes up.(As shown in attached screenshot). Could you refer me to a test case model which has wake modelling visualisation for a variable wind speed(doesnt matter if single turbine or multiple ? I seem to have issues in some test cases in the glue codes. 
 
[image]

Prithvi2029, 2022-05-27:
Also, I have made the changes to the file as you suggested, Please review the pull request. 
Link to request - https://github.com/OpenFAST/r-test/pull/59

jjonkman, 2022-05-27:
Regarding your new error, it appears that the InflowWind input file you are using is also out of date relative to OpenFAST v3.1.  An InflowWind input file is not included in the 5MW_Land_ModeShapes test, but you can find a properly formatted InflowWind input file here: https://github.com/OpenFAST/r-test/blob/main/glue-codes/openfast/5MW_Baseline/NRELOffshrBsline5MW_InflowWind_12mps.dat.

Regarding PR #59, you'll also need to add `CompAA` and `AA_InputFile` to the AeroDyn v15 input file for compatibility with the newest version of AeroDyn.

I'm not sure I understand your question regarding wake visualization.

Best regards,

Prithvi2029, 2022-05-27:
Hello @jjonkman , 
Thank you for your suggestions. I have added those 2 lines as we discussed. But it's now throwing an error with missing 'VFlowAng' in file - 5MW_Baseline/NRELOffshrBsline5MW_InflowWind_Steady13mps.dat. (the last screenshot is attached)

By wake visualization, I meant that , if you can share with me the test case which shows wake effects in wind turbines such as shown in the image below. I am trying to apply some algorithms related to the wake model and visualize as the wind speed and direction change. Thereby use yaw control.  
[image]

Also, I have added the Discon.dll file with the mentioned commit, Please review - https://github.com/OpenFAST/r-test/pull/60

[image]

jjonkman, 2022-05-27:
I responded regarding your InflowWind input file in my prior post.

We purposely don't distribute precompiled DISCON binary files for Windows in the r-test, which is otherwise generally applicable for any operating system.  Instead, these are provided separately here: https://github.com/OpenFAST/openfast/releases/download/v3.1.0/DISCON.DLLS.zip.  Please close PR #60.

Wake effects are computed within FAST.Farm, not standalone OpenFAST.  The r-test for FAST.Farm is available here: https://github.com/OpenFAST/r-test/tree/main/glue-codes/fast-farm.

Best regards,

Prithvi2029, 2022-05-27:
After running LEFInflow which is from r-test of Fast farm, its missing the DISCON_WT1.dll file(screenshot 1). Do I have to edit the dll file to same name as missing file from the zip file from the above link you shared ? and even after editing the dll file name , FAST.farm isnt working. (screenshot 2 after I addded the DISCON.dll file from zip folder and changed the name to DISCON_WT1.dll)

[image]

[image]

jjonkman, 2022-05-27:
Yes, that is correct.  The original _DISCON.dll_ for the NREL 5-MW baseline wind turbine must be renamed to _DISCON_WT1.dll_ for this test (no other change needs to be made, just the renaming).

Please note, however, that it appears that you are running standalone OpenFAST rather than FAST.Farm.  To run FAST.Farm, you must use a compiled binary executable of FAST.Farm and the FAST.Farm input file (_LESinflow.fstf_ for the LESinflow test).

Prithvi2029, 2022-05-28:
Thank you for your patience and guidance, I was able to run the LES Inflow model. I appreaciate your inputs !

closed 2022-10-25 by ebranlard