OpenFAST/r-test #64: issue with running MD V2 farm branch

huzaifapolimi, 2022-07-06:
I am trying to use the new moordyn V2 with added bending stiffness but unfortunately there's an error which i am having a hard time figuring it out.  

**FAST_InitializeAll:FAST_Init:FAST_ReadPrimaryFile:Invalid logical input for file
".\5MW_OC4Semi_WSt_WavesWN.fst" occurred while trying to read SumPrint.**

I am aware of the new .fst file with added environmental conditions  which i am currently trying to use. If anyone has had the same issue and been able to solve it . It would be helpful if you shared your approach.

huzaifapolimi, 2022-07-11:
Dear @jjonkman 
Thanks for the help.. glad to be finally using the Moordyn with bending stiffness. its great!!!

andrew-platt, 2022-11-23:
MoorDyn V2 is now in OpenFAST as of v 3.3.0.

I think we can close this issue at this point.

closed 2022-11-23 by andrew-platt