OpenFAST/r-test #66: invalid logical input to read SumPrint

rmaederzii, 2022-07-21:
Dear all,

I have a problem in running the OPENfast with FFTest_WT1.fst input files. I receive an error on reading the input files "FFTest_WT1.fst". The error is as follows:
Invalid logical input for file ".\FFTest_WT1.fst" occurred while trying to read SumPrint.

FAST.Farm-v3.0.0
Compile Info:
 - Compiler: Intel(R) Fortran Compiler 1900
 - Architecture: 64 bit
 - Precision: single
 - OpenMP: No
 - Date: Jun 23 2021
 - Time: 21:41:28
Execution Info:
 - Date: 07/21/2022
 - Time: 09:22:15+0430

 Heading of the FAST.Farm input file:
   Sample FAST.Farm input file
Running AWAE.
Running WakeDynamics.
Running FASTWrapper (v1.00.00, 7-Feb-2017).
Running OpenFAST.
OpenFAST input file heading:
    FAST Certification Test #26: NREL 5.0 MW Baseline Wind Turbine (Onshore)

Farm_Initialize:Farm_ValidateInput:OutFmt produces a column width of 10 instead of 20 characters.
Farm_Initialize:For efficiency, NumPlanes has been reduced to the number of time steps (16).
Farm_Initialize:T1:Farm_InitFAST:FWrap_Init:FAST_InitializeAll:FAST_Init:FAST_ReadPrimaryFile:Inva
lid logical input for file ".\FFTest_WT1.fst" occurred while trying to read SumPrint.

 Aborting FAST.Farm.

I would greatly appreciate if you could help me with this problem.

Best regards,
Maede

jjonkman, 2022-07-21:
Dear @rmaederzii,

My guess is that you are using input files that are formatted for a different version of FAST.Farm / OpenFAST than you are simulating with.  It looks like you are running FAST.Farm from the release of OpenFAST v3.0.  Are you input files for that version?  Perhaps try upgrading to the release of OpenFAST v3.2 with the r-test for v3.2?

Alternatively, you can enable the `Echo` option to debug errors in the input file processing and review the documentation on input file changes between different versions of the software: https://openfast.readthedocs.io/en/main/source/user/api_change.html.

Best regards,

rmaederzii, 2022-07-22:
Dear @jjonkman
i'm just started 
can you give me a sample input for version 3.0 to get startet?

Best regards

rmaederzii, 2022-07-23:
dear @rafmudaf 
 I receive an error on reading the dynamic library . The error is as follows:
dynamic library .\../5MW_Baseline/ServoData/DISCON_WT1.dll could not be loaded. Check that the
file exists in the specified location and that it is compiled for 64-bit applications.
FAST.Farm-v3.0.0
Compile Info:
 - Compiler: Intel(R) Fortran Compiler 1900
 - Architecture: 64 bit
 - Precision: single
 - OpenMP: No
 - Date: Jun 23 2021
 - Time: 21:41:28
Execution Info:
 - Date: 07/23/2022
 - Time: 10:32:26+0430

 Heading of the FAST.Farm input file:
   Sample FAST.Farm input file
Running AWAE.
Running InflowWind.

   Reading a 101x35 grid (1000 m wide, 5 m to 345 m above ground) with a characteristic wind
   speed of 9.243 m/s. This full-field file was generated by TurbSim (v2.00.07a-bjj, 14-Jun-2016)
   on 24-Sep-2019 at 16:18:09.

   Processed 2000 time steps of 10-Hz full-field data (period of 200 seconds).
Running WakeDynamics.
Running FASTWrapper (v1.00.00, 7-Feb-2017).
Running OpenFAST.
OpenFAST input file heading:
    FAST Certification Test #26: NREL 5.0 MW Baseline Wind Turbine (Onshore)

Running ElastoDyn.
Nodal outputs section of ElastoDyn input file not found or improperly formatted.
Running AeroDyn.
AD15 Nodal Outputs: Nodal output section of AeroDyn input file not found or improperly formatted.
Skipping nodal outputs.
Running InflowWind.
Running ServoDyn.
Running ServoDyn Interface for Bladed Controllers (using Intel Visual Fortran for Windows, ).

Farm_Initialize:Farm_ValidateInput:OutFmt produces a column width of 10 instead of 20 characters.
Farm_Initialize:For efficiency, NumPlanes has been reduced to the number of time steps (134).
Farm_Initialize:T1:Farm_InitFAST:FWrap_Init:FAST_InitializeAll:SrvD_Init:BladedInterface_Init:The
dynamic library .\../5MW_Baseline/ServoData/DISCON_WT1.dll could not be loaded. Check that the
file exists in the specified location and that it is compiled for 64-bit applications.

 Aborting FAST.Farm.
I would greatly appreciate if you could help me with this problem.

Best regards,
Maede