OpenFAST/r-test #68: Grid too small in Y direction.

rmaederzii, 2022-08-17:
Dear all,
i want to use r-test-3.0.0 to simulate a wind farm with power of 50 MW in the form of two rows with 5 turbines in each row,but every time i add turbines i encounter this error.

FAST.Farm-v3.0.0
Compile Info:
 - Compiler: Intel(R) Fortran Compiler 1900
 - Architecture: 64 bit
 - Precision: single
 - OpenMP: No
 - Date: Jun 23 2021
 - Time: 21:41:28
Execution Info:
 - Date: 08/17/2022
 - Time: 10:07:33+0430

 Heading of the FAST.Farm input file:
   Sample FAST.Farm input file
Running AWAE.
Running InflowWind.

   Reading a 101x35 grid (1000 m wide, 5 m to 345 m above ground) with a characteristic wind
   speed of 9.243 m/s. This full-field file was generated by TurbSim (v2.00.07a-bjj, 14-Jun-2016)
   on 24-Sep-2019 at 16:18:09.

   Processed 2000 time steps of 10-Hz full-field data (period of 200 seconds).

Farm_Initialize:Farm_ValidateInput:OutFmt produces a column width of 10 instead of 20 characters.
Farm_Initialize:For efficiency, NumPlanes has been reduced to the number of time steps (134).
Farm_Initialize:InflowWind_CalcOutput:CalcOutput:IfW_FFWind_CalcOutput [position=(4810.4, 1510,
345)  in wind-file coordinates]: FF wind array boundaries violated: Grid too small in Y
direction. Y=1510; Y boundaries = [-500, 500]

 Aborting FAST.Farm.

I would greatly appreciate if you could help me with this problem.

Best regards,
Maede

jjonkman, 2022-08-17:
Dear @rmaederzii,

It sounds like you've defined a larger low- or high-resolution wind domain in FAST.Farm without updating the TurbSim data accordingly.  You must define the ambient wind data in TurbSim consistent with how the low- and high-resolution domains are defined in FAST.Farm.  See the Modeling Guidance section of the online FAST.Farm documentation for more information: https://openfast.readthedocs.io/en/main/source/user/fast.farm/ModelGuidance.html.

Best regards,

rmaederzii, 2022-08-20:
Dear all,
i want to use r-test-3.0.0 to simulate a wind farm with power of 50 MW in the form of two rows with 5 turbines in each row,but every time i add turbines i encounter this error.
FAST.Farm-v3.0.0
Compile Info:
 - Compiler: Intel(R) Fortran Compiler 1900
 - Architecture: 64 bit
 - Precision: single
 - OpenMP: No
 - Date: Jun 23 2021
 - Time: 21:41:28
Execution Info:
 - Date: 08/20/2022
 - Time: 09:35:31+0430

 Heading of the FAST.Farm input file:
   Sample FAST.Farm input file
Running AWAE.
Running InflowWind.

   Reading a 7x800 grid (170 m wide, 5 m to 175 m above ground) with a characteristic wind speed
   of 8.092 m/s. This full-field file was generated by TurbSim (v2.00.07a-bjj, 14-Jun-2016) on
   20-Aug-2022 at 09:09:25.

   Processed 2000 time steps of 10-Hz full-field data (period of 200 seconds).

Farm_Initialize:Farm_ValidateInput:OutFmt produces a column width of 10 instead of 20 characters.
Farm_Initialize:For efficiency, NumPlanes has been reduced to the number of time steps (134).
Farm_Initialize:InflowWind_CalcOutput:CalcOutput:IfW_FFWind_CalcOutput [position=(3681.5, 334,
4995)  in wind-file coordinates]: FF wind array boundaries violated. Grid too small in Z
direction (Z=4995 m is above the grid).

 Aborting FAST.Farm.
I would greatly appreciate if you could help me with this problem.

Best regards,
Maede

rmaederzii, 2022-08-20:
i set the numgrid znin the *.fstf to be same as in the turbsim file.but still this error occure

jjonkman, 2022-08-22:
Dear @rmaederzii,

Your new error sounds very similar to the original error you posted and implies that the TurbSim and FAST.Farm domains are not compatible.  It is not just the number of grid points that matters in X, Y, and Z, but the full extent of the grid.  And your new error implies your FAST.Farm model is set up quite strangely given that a node is located Z = 4995 m.

Again, see the Modeling Guidance section of the online FAST.Farm documentation for more information: https://openfast.readthedocs.io/en/main/source/user/fast.farm/ModelGuidance.html.

Best regards,

rmaederzii, 2022-08-24:
Dear all,
i want to use r-test-3.0.0 to simulate a wind farm with power of 25 MW in the form of 1 rows with 5 turbines in one row,but every time i add turbines i encounter this error.
FAST.Farm-v3.0.0
Compile Info:
 - Compiler: Intel(R) Fortran Compiler 1900
 - Architecture: 64 bit
 - Precision: single
 - OpenMP: No
 - Date: Jun 23 2021
 - Time: 21:41:28
Execution Info:
 - Date: 08/24/2022
 - Time: 10:34:34+0430

 Heading of the FAST.Farm input file:
   Sample FAST.Farm input file
Running AWAE.
Running InflowWind.

   Reading a 101x35 grid (1000 m wide, 5 m to 345 m above ground) with a characteristic wind
   speed of 9.243 m/s. This full-field file was generated by TurbSim (v2.00.07a-bjj, 14-Jun-2016)
   on 24-Sep-2019 at 16:18:09.

   Processed 2000 time steps of 10-Hz full-field data (period of 200 seconds).
Running WakeDynamics.
Running FASTWrapper (v1.00.00, 7-Feb-2017).
Running OpenFAST.
OpenFAST input file heading:
    FAST Certification Test #26: NREL 5.0 MW Baseline Wind Turbine (Onshore)

Running ElastoDyn.
Nodal outputs section of ElastoDyn input file not found or improperly formatted.
Running AeroDyn.
AD15 Nodal Outputs: Nodal output section of AeroDyn input file not found or improperly formatted.
Skipping nodal outputs.
Running InflowWind.
Running ServoDyn.
Running ServoDyn Interface for Bladed Controllers (using Intel Visual Fortran for Windows, ).
Using legacy Bladed DLL interface.
Running OpenFAST.
OpenFAST input file heading:
    FAST Certification Test #26: NREL 5.0 MW Baseline Wind Turbine (Onshore)

Running ElastoDyn.
Nodal outputs section of ElastoDyn input file not found or improperly formatted.
Running AeroDyn.
AD15 Nodal Outputs: Nodal output section of AeroDyn input file not found or improperly formatted.
Skipping nodal outputs.
Running InflowWind.
Running ServoDyn.
Running ServoDyn Interface for Bladed Controllers (using Intel Visual Fortran for Windows, ).
Using legacy Bladed DLL interface.
Running OpenFAST.
OpenFAST input file heading:
    FAST Certification Test #26: NREL 5.0 MW Baseline Wind Turbine (Onshore)

Running ElastoDyn.
Nodal outputs section of ElastoDyn input file not found or improperly formatted.
Running AeroDyn.
AD15 Nodal Outputs: Nodal output section of AeroDyn input file not found or improperly formatted.
Skipping nodal outputs.
Running InflowWind.
Running ServoDyn.
Running ServoDyn Interface for Bladed Controllers (using Intel Visual Fortran for Windows, ).
Using legacy Bladed DLL interface.
Running OpenFAST.
OpenFAST input file heading:
    FAST Certification Test #26: NREL 5.0 MW Baseline Wind Turbine (Onshore)

Running ElastoDyn.
Nodal outputs section of ElastoDyn input file not found or improperly formatted.
Running AeroDyn.
AD15 Nodal Outputs: Nodal output section of AeroDyn input file not found or improperly formatted.
Skipping nodal outputs.
Running InflowWind.
Running ServoDyn.
Running ServoDyn Interface for Bladed Controllers (using Intel Visual Fortran for Windows, ).
Using legacy Bladed DLL interface.
Running OpenFAST.
OpenFAST input file heading:
    FAST Certification Test #26: NREL 5.0 MW Baseline Wind Turbine (Onshore)

Running ElastoDyn.
Nodal outputs section of ElastoDyn input file not found or improperly formatted.
Running AeroDyn.
AD15 Nodal Outputs: Nodal output section of AeroDyn input file not found or improperly formatted.
Skipping nodal outputs.
Running InflowWind.
Running ServoDyn.
Running ServoDyn Interface for Bladed Controllers (using Intel Visual Fortran for Windows, ).
Using legacy Bladed DLL interface.
 Time: 0 of 400 seconds.

T1:FARM_InitialCO:FWrap_t0:FAST_Solution0:CalcOutputs_And_SolveForInputs:SolveOption2:SrvD_CalcOut
put:DLL_controller_call:Running with torque and pitch control of the NREL offshore 5MW baseline
wind turbine from DISCON.dll as written by J. Jonkman of NREL/NWTC for use in the IEA Annex XXIII
OC3 studies.
T2:FARM_InitialCO:FWrap_t0:FAST_Solution0:CalcOutputs_And_SolveForInputs:SolveOption2:SrvD_CalcOut
put:DLL_controller_call:Running with torque and pitch control of the NREL offshore 5MW baseline
wind turbine from DISCON.dll as written by J. Jonkman of NREL/NWTC for use in the IEA Annex XXIII
OC3 studies.
T3:FARM_InitialCO:FWrap_t0:FAST_Solution0:CalcOutputs_And_SolveForInputs:SolveOption2:SrvD_CalcOut
put:DLL_controller_call:Running with torque and pitch control of the NREL offshore 5MW baseline
wind turbine from DISCON.dll as written by J. Jonkman of NREL/NWTC for use in the IEA Annex XXIII
OC3 studies.
T4:FARM_InitialCO:FWrap_t0:FAST_Solution0:CalcOutputs_And_SolveForInputs:SolveOption2:SrvD_CalcOut
put:DLL_con

 Aborting FAST.Farm.
best regards.

jjonkman, 2022-08-24:
Dear @rmaederzii.

It appears that there are many messages written regarding the calling of the controller that the character string hits the limit on number of characters permitted.  So, the message associated with the actual error causing FAST.Farm to abort is not shown.  I would suggest temporarily disabling the controller to identify the actual error.  You can temporarily disable the controller by setting CompServo = 0 in the OpenFAST primary (.fst) input file and by setting `GenDOF` = `YawDOF` = False in ElastoDyn.

Best regards,

rmaederzii, 2022-08-25:
Dear @jjonkman 
with what parameters schould we define the Domain in turbsim to make sure that no points fall outside the Domain?

jjonkman, 2022-08-25:
Dear @rmaederzii,

The size of the TurbSim domain in the Y and Z directions is determined by TurbSim inputs `HubHt`, `GridHeight`, and `GridWidth`.  The size of the TurbSim domain in the X direction can be infinite (due to its periodicity) when `UsableTime` = "ALL".

Defining turbulent wind inflow for FAST.Farm using TurbSim without introducing excessive interpolation is not straightforward.  This is why we encourage you to follow the provided modeling guidance (linked above) and associated Python scripts.  NREL is actively working on improvements to FAST.Farm and the associated Python scripts to make the process easier.

Best regards,

rmaederzii, 2022-08-28:
Dear @jjonkman 
how can i calculate the efficiency and power generate of the farm?

Best regards,

jjonkman, 2022-08-29:
Dear @rmaederzii,

Is your question specifically tied to FAST.Farm?  You can output the power of each wind turbine in the farm via the associated output from OpenFAST for each wind turbine.  I'm not sure how you would define wind farm efficiency, but it is quite common to assess how the power production of downstream wind turbines in the farm compare to the upstream wind turbines, which can certainly be compare when outputting the power of each turbine.

Best regards,

closed 2022-12-13 by andrew-platt