OpenFAST/r-test #69: fast on mpi

huzaifapolimi, 2022-09-20:
Hi,
I am facing an issue with executing openfast on mpi.  unfortunately it is unable to find the .fst file even when located in the very same directory that its being launched in. I have checked the launch script very carefully multiple times. there seems to be no errors in it. 
error- 
FAST_InitializeALL:FAST_Init:FAST_ReadPrimaryFile:OpenFInputFile:The input file,"./" , was not found.

Is fast compatible with mpi?

bjonkman, 2022-09-20:
What is the command you are using to call OpenFAST? It looks like you may have a space (or some other Fortran record-break character) between the path name and the file name because it is parsing the file name as `./`.

huzaifapolimi, 2022-09-20:
The command is just the path to the executable and then followed by the input file(.fst).
#!/bin/bash      #use bash as command interpreter
#$ -cwd          #currentWorkingDirectory
#$ -N trial_5MW  # jobName
#$ -j y         #merges output and errors
#$ -S /bin/bash  #scriptinng language
#$ -q mecc4.q   # queueName
#$ -pe mpi 1    #Cpu number
source $HOME/bashrc_openfast
###specify the executable ...
/global-scratch/bulk_pool/hsyed/trial/openfast/install/bin/openfast  Semisub.fst
echo End Parallel Run

huzaifapolimi, 2022-09-20:
FAST_InitializeAll:FAST_Init:FAST_ReadPrimaryFile:OpenFInpFile:The input file,
 "/global-scratch/bulk_pool/hsyed/testing/glue-codes/openfast/5MW_OC4Semi_WSt_WavesWN/Semisub.fst
"
 , was not found.

 OpenFAST encountered an error during module initialization.
  Simulation error level: FATAL ERROR

  Aborting OpenFAST.

End Parallel Run

and this is the exact error popping up.

bjonkman, 2022-09-20:
I'm very surprised the code would give you the error message listing the whole path to the input file if you called it without any path specified (as is shown in the command you said you were using).

Can you check that the name of the .fst file you are trying to open appears with the `ls` command (and that you can open/read it)? Use the _exact_ argument you are using for the file name argument when you call OpenFAST. It's possible there are non-printable characters in your argument or in the path name, so you need to be exact.

huzaifapolimi, 2022-09-21:
Yes the .fst file appears with the ls command. and i am able to read it as well. but unfortunately the error still continues to be the same.
[image]

andrew-platt, 2024-06-29:
Closing this after 2 years of no activity.

closed 2024-06-29 by andrew-platt