OpenFAST/r-test #90: Aerodynamic calculation using AeroDyn standalone

hefe2345, 2023-02-22:
I am trying to perform the aerodynamic calculation for the wind turbine blade. I use AeroDyn standalone to perform the analysis. While giving the input, I got the error "error reading NumAlf". Is it necessary to have an equal number of data lines for the aerodynamic coefficient table and aerofoil data file(I consider the NREL input dataset for calculation with 400 lines of airfoil data and 146 lines of aerodynamic coefficient data)?
Any suggestions you can give me will be constructive.

jjonkman, 2023-02-22:
Dear @hefe2345,

I'm not sure I fully understand your question (are you asking if the number of lines in the airfoil coefficients table in separate airfoil data files must be the same?).  Regardless, it sounds like your specific issue is related to the format of you airfoil data file.  As with any input file formatting problem, I suggest enabling the `Echo` option to debug.

Best regards,

hefe2345, 2023-02-22:
Dear jjonkman,
I tried both cases (separate aerofoil data files and together in the same file), and I got a different error in both cases. While using an additional file, the error was in NumTabs, and the later case was in NumAlf. Screenshots of both cases are attached.
[image]
[image]
Echo is also True in both case.
Thank you

ebranlard, 2023-02-22:
Hi @hefe2345 
I think you are running into an issue where your input files are not at the same version of your executable. The version of the 
One way to make sure of that is to go to a given folder in the r-test, and select a given tag (same dropdown menu as for selecting branches), here's an example of file for the tag/version 3.4.1: 
https://github.com/OpenFAST/r-test/blob/v3.4.1/modules/aerodyn/ad_BAR_SineMotion/OpenFAST_BAR_00_AeroDyn15.dat

In your case, it seems you are running a very old version of AeroDyn driver. You'll have to compile a new version from scratch. 
See: https://openfast.readthedocs.io/en/dev/source/install/index.html#compile-from-source
On Windows, the easiest is to install intel one API, and Visual Studio, and then compile using the visual studio solution for the aerodyn driver. 

I hope that helps,

closed 2024-06-29 by andrew-platt