OpenFAST/r-test #128: SWRT tests cases with tail fin do not match FAST7 results

andrew-platt, 2024-06-29:
The _SWRT_ cases now include a new tail fin model that theoretically is similar to the _FAST7_ cases (see PR #126).

Comparing the results for the cases with _FAST7_ shows some notable differences.  There have been significant improvements to the _AeroDyn_ that may account for the differences.  Someone should double check this though.

[image]
Example from the `SWRT_YFree_VS_EDC01` case (`Test16.out` from the _FAST7_ CertTest).  Note that `SubAxInd` has been turned off in the _FAST7_ run since `TFinIndMod=1` is not available in the current dev branch of OpenFAST (3d476c595).

@abhineet-gupta may have some insights here.

abhineet-gupta, 2024-07-16:
I did some digging and reduced the problem to a very simple case of steady wind, all flexible and turbine degrees of freedom switched off; but I still couldn't match the tail fin forces. OpenFAST gives expected results but Fast V7 does not.

[Compare.pdf](https://github.com/user-attachments/files/16257155/Compare.pdf)

Not sure what's going on with Fast V7 here (or am I missing something) but will look into it and discuss it with @jjonkman if I can't find a solution.

andrew-platt, 2024-08-26:
After further discussion, we concluded that FAST7 is incorrect.  Since we are not going to update FAST7, we are considering this as solved and will close the issue.

closed 2024-08-26 by andrew-platt