OpenFAST/r-test #130: Lift coefficients in BAR airfoil files

bjonkman, 2024-07-24:
While debugging some differences in a new method of calculating default unsteady airfoil coefficients, I noticed some unusual characteristics in the lift coefficients in several of the aerodyn/BAR_Baseline files (1-6). 

./modules/aerodyn/BAR_Baseline/Airfoils/BAR0_AeroDyn15_Polar_01.dat
./modules/aerodyn/BAR_Baseline/Airfoils/BAR0_AeroDyn15_Polar_02.dat
./modules/aerodyn/BAR_Baseline/Airfoils/BAR0_AeroDyn15_Polar_03.dat
./modules/aerodyn/BAR_Baseline/Airfoils/BAR0_AeroDyn15_Polar_04.dat
./modules/aerodyn/BAR_Baseline/Airfoils/BAR0_AeroDyn15_Polar_05.dat
./modules/aerodyn/BAR_Baseline/Airfoils/BAR0_AeroDyn15_Polar_06.dat

I am not sure what those local minima just below alpha=0 are:
 [image]

Other aerodyn/BAR_Baseline airfoils look more normal.  For example

./modules/aerodyn/BAR_Baseline/Airfoils/BAR0_AeroDyn15_Polar_07.dat
./modules/aerodyn/BAR_Baseline/Airfoils/BAR0_AeroDyn15_Polar_08.dat
./modules/aerodyn/BAR_Baseline/Airfoils/BAR0_AeroDyn15_Polar_09.dat

Here are values for 7-9: 
[image]

If those local minima are expected, we will have to look into some different methods for computing values for `c_lalpha` and `c_nalpha`. Also, I'm not sure how the UA models will handle it. 

Let me know what you think.

ebranlard, 2024-07-24:
Good catch, probably the yaml airfoil data at 10-13% span has a bit of a reversed lift curve and then the linear interpolation between airfoils does that. I'm curious to know what Pietro finds out :)

ptrbortolotti, 2024-07-26:
hello, I looked into this and I have some thoughts:

- BAR0 is an older iteration of the rotor designs developed within the Big Adaptive Rotor project. BAR0 evolved into BAR-USC, which is available here https://github.com/NREL/BAR_Designs
- Both BAR0 and BAR-USC use the thick airfoil SNL-FFA-W3-500, whose coordinates are listed here https://github.com/NREL/BAR_Designs/blob/10e11353208520a052622490693099e1e5404be0/BAR_USC/BAR_USC.yaml#L520 
[image]

- The BAR-USC design is described in this journal article https://doi.org/10.5194/wes-6-1277-2021. At the time of publishing, the polars of BAR-USC were computed with XFoil
- Later on, we used the 2D CFD solver HAMD2D, which is described in this journal article https://doi.org/10.5194/wes-7-603-2022, to recompute the polars. These new polars are the ones listed here https://github.com/NREL/BAR_Designs/blob/10e11353208520a052622490693099e1e5404be0/BAR_USC/BAR_USC.yaml#L525
- The lift coefficient of this airfoil looks funky
[image]
- This is not the first thick airfoil where I see funky Cl
- The interpolation between this lift curve and the lift of the cylinder at blade root creates funkier lift curves. Here it might be best to simply turn off the unsteady airfoil aero models
- In the RAAW project we struggled with poor performance of some of the UAMod models https://openfast.readthedocs.io/en/main/source/user/aerodyn/theory_ua.html. We ended up toggling between UAMod 3 and 4, more details available here https://doi.org/10.5194/wes-2023-166 
- NREL and UMass have just submitted a proposal for a new research project named STABLE to the Wind Energy Technologies Office of DOE. Among other things, STABLE will review, verify, and validate the various models behind the UAMod flags. @ebranlard and his students will lead this work

ebranlard, 2024-07-26:
Thanks Pietro. I think an important question is how much should we trust the lift curve of these thick airfoils, and is the spanwise interpolation a decent method (most people use it even though we know it's not ideal, but often preferred to jumps of polars like in the NREL 5 MW) or should we come up with another method to avoid interpolating polars (might be hard).