OpenFAST/r-test #153: TSinflow

aothmane-control, 2025-05-20:
The example TSinflow for FAST.farm seems to be outdated. Running openfast in the folder of the examples gives the results
`openfast FAST.Farm.fstf 

 **************************************************************************************************
 OpenFAST

 Copyright (C) 2025 National Renewable Energy Laboratory
 Copyright (C) 2025 Envision Energy USA LTD

 This program is licensed under Apache License Version 2.0 and comes with ABSOLUTELY NO WARRANTY.
 See the "LICENSE" file distributed with this software for details.
 **************************************************************************************************

 OpenFAST-v4.0.4
 Compile Info:
  - Compiler: GCC version 13.3.0
  - Architecture: 64 bit
  - Precision: single
  - OpenMP: No
  - Date: Apr 29 2025
  - Time: 14:26:26
 Execution Info:
  - Date: 05/20/2025
  - Time: 17:27:46+0200

 OpenFAST input file heading:
     FAST.Farm input file, using two turbines separated by 3D downstream and an offset of 30m,
     with a turbulent inflow given by one turbsim box

 FAST_InitializeAll:FAST_Init:FAST_ReadPrimaryFile:Invalid numerical input for file
 "FAST.Farm.fstf" occurred while trying to read DT.

 OpenFAST encountered an error during module initialization.
  Simulation error level: FATAL ERROR

  Aborting OpenFAST.
`

andrew-platt, 2025-05-20:
Which branch of the regression test are you running?  Did you update it to 4.0.4 as well?

jjonkman, 2025-05-20:
Dear @aothmane-control,

The problem seems to be in how you are running this test.  You should call the FAST.Farm executable (e.g., _FAST.Farm_x64.exe_) instead of the OpenFAST executable.

Best regards,